PaulStoffregen/teensy_loader_cli #5: Ability to flash multiple teensys?

mgcrea, 2015-03-25:
I have an Ubuntu box with 5 teensy boards connected, I'm trying to flash them with an .hex file, but it looks like I can only flash a single one. Any idea on how to solve that? Would it be possible to add an optionnal parameter to select the device?

PaulStoffregen, 2015-03-25:
You could try adding this to the soft reboot code.  But how to distinguish which device?  I can't get involved in this.

mgcrea, 2015-03-25:
I'll try to find something, thanks.

Here is the output with debug enabled:

```
$ sudo teensy_loader_cli --mcu=mk20dx256 build-teensy31/teensy-cube-make.hex -v -s
Teensy Loader, Command Line, Version 2.0
Read "build-teensy31/teensy-cube-make.hex": 41568 bytes, 15.9% usage

Searching for USB device:
bus "002", device "001" vid=1D6B, pid=0003
bus "001", device "029" vid=16C0, pid=0483
bus "001", device "010" vid=16C0, pid=0483
bus "001", device "009" vid=16C0, pid=0483
bus "001", device "008" vid=16C0, pid=0483
bus "001", device "007" vid=1A40, pid=0101
bus "001", device "006" vid=05E3, pid=0608
bus "001", device "005" vid=16C0, pid=0483
bus "001", device "004" vid=2341, pid=0043
bus "001", device "003" vid=8087, pid=07DC
bus "001", device "002" vid=05E3, pid=0608
bus "001", device "001" vid=1D6B, pid=0002

Searching for USB device:
bus "002", device "001" vid=1D6B, pid=0003
bus "001", device "029" vid=16C0, pid=0483
Soft reboot performed
Waiting for Teensy device...
 (hint: press the reset button)

Searching for USB device:
bus "002", device "001" vid=1D6B, pid=0003
bus "001", device "029" vid=16C0, pid=0483
bus "001", device "010" vid=16C0, pid=0483
bus "001", device "009" vid=16C0, pid=0483
bus "001", device "008" vid=16C0, pid=0483
bus "001", device "007" vid=1A40, pid=0101
bus "001", device "006" vid=05E3, pid=0608
bus "001", device "005" vid=16C0, pid=0483
bus "001", device "004" vid=2341, pid=0043
bus "001", device "003" vid=8087, pid=07DC
bus "001", device "002" vid=05E3, pid=0608
bus "001", device "001" vid=1D6B, pid=0002

Searching for USB device:
bus "002", device "001" vid=1D6B, pid=0003
bus "001", device "010" vid=16C0, pid=0483
bus "001", device "009" vid=16C0, pid=0483
bus "001", device "008" vid=16C0, pid=0483
bus "001", device "007" vid=1A40, pid=0101
bus "001", device "006" vid=05E3, pid=0608
bus "001", device "005" vid=16C0, pid=0483
bus "001", device "004" vid=2341, pid=0043
bus "001", device "003" vid=8087, pid=07DC
bus "001", device "002" vid=05E3, pid=0608
bus "001", device "001" vid=1D6B, pid=0002

Searching for USB device:
bus "002", device "001" vid=1D6B, pid=0003
bus "001", device "030" vid=16C0, pid=0478
Found HalfKay Bootloader
Read "build-teensy31/teensy-cube-make.hex": 41568 bytes, 15.9% usage
Programming.........................................
Booting
```