PaulStoffregen/teensy_loader_cli #9: Unable to flash Teensy LC

expipiplus1, 2015-04-28:
I have a new Teensy 3.1 and a Teensy LC. I'm able to flash the Teensy 3.1 with teensy_loader_cli without a problem, however I've had no success in flashing the Teensy LC.

Just to make sure, here's what I'm running:

For the Teensy 3.1: `sudo ./teensy_loader_cli --mcu=mk20dx256 -w -v blink_slow_Teensy31.hex`
For the Teensy LC: `sudo ./teensy_loader_cli --mcu=mkl26z64 -w -v blink_slow_TeensyLC.hex`

When trying to flash the LC, `usb_control_msg` returns -32 (EPIPE) in `teensy_write`.
I tried this from three linux installations (Ubuntu 14.10, Ubuntu 15.04 and NixOS) with identical results.

I am able to use Teensyduino on linux to flash both boards.
I am able to use the graphical teensy loader on OSX to flash both boards.
It's just the cli loader which fails on linux for the LC board.

I'm more than happy to provide you with any more information. How would be best for me to proceed in debugging this? Is there a specification for programming the teensy boards?

des09, 2015-05-17:
experiencing same issue.

PaulStoffregen, 2015-05-21:
I've committed a fix.  Please try the latest code and let me know if it works for you?

closed 2016-11-09 by PaulStoffregen