PaulStoffregen/teensy_loader_cli #13: Please add version tags

kkaempf, 2015-09-22:
With version tags, github provides downloadable tarballs, making it easier for Linux package maintainers to package your code.

blackketter, 2016-03-09:
Would also be great to bump the version number in the code so it's available from the command line.

ticky, 2017-01-07:
Yep, just encountered some confusion because the build in Homebrew (for macOS) is behind the version on the website, because the version in Homebrew is the latest “version” but the one on the website is newer. Version numbers are a must!!!

hmaarrfk, 2019-11-09:
Bump for this request.

I just packaged it up for conda-forge and I wanted to give the latest source a name. I just had to name it `2.1.1`, but it really isn't