PaulStoffregen/teensy_loader_cli #18: Use getopt for option parsing

dagon666, 2016-01-04:
Hi,

This is more of a question than an issue. How do you feel about using getopt() for option parsing ? It would make the code much cleaner and easier to maintain. It would have to be compiled under cygwin for Windows though. The use of cygwin on Windows would allow to use a unified USB communication library in the code as well, which in my opinion would be generally beneficial.

Let me know what do you think. I already have a getopt change ready. I could provide some more changes if you agree that this is a good idea.

PaulStoffregen, 2016-01-06:
Kinda seems like overkill.  There's only 39 lines now for parse_options().

I really do want to preserve the ability to build with MinGW.  Anything that forces use of Cygwin really needs to be a very compelling feature.

dagon666, 2016-01-07:
There's libusb-win32 for MingW http://www.libusb.org/wiki/libusb-win32 which would allow to preserve a single portable implementation ang get rid of the Windows specific USB code.

The current option parsing code maybe is pretty short but at the same time it's pretty sloppy. Being precise you should take into account the parse_flag() function as well. Have a look on my branch. That change is already available: dagon666/teensy_loader_cli@e1800804de3b7fb23efd8b5d34c9c4e11cdba2fd

I can create the PR for this change and take care about usb code as well.

PaulStoffregen, 2016-01-07:
> There's libusb-win32 for MingW

Can it properly detach the Windows HID driver, like it does for RawHID on Linux?  When I looked years ago, it couldn't.  Without that feature, it's worthless.