PaulStoffregen/teensy_loader_cli #23: Outdated windows libraries?

Dimpl, 2016-04-27:
I tried to compile on windows, and am receiving an error that 'ddk/hidsdi.h' does not exist. (I do, however have 'ddk/hidclass.h'). Is this file outdated? I tried to download missing files (and dependencies) [here](https://sourceforge.net/u/earnie/winapi/winapi/ci/f9fa22a4ecce88ad12a8371a40f00ef702ec3e1b/tree/include/ddk/), but more errors ensued, which I'm guessing is due to file incompatibility.

PaulStoffregen, 2016-04-27:
Those are part of the WDK.  Why Microsoft doesn't include them with their normal tools is beyond me.

FWIW, I have *never* compiled this using Visual Studio or any other Microsoft tool.  I do all my building on Linux using mingw as a cross compiler.

If you use Ubuntu Linux and install the mingw tools with apt-get, it will "just work" and produce a good .exe file, which runs when copied to a Windows computer.  That's how I do it.  So I really can't help with how to do it actually on Windows.  I pretty much never use Windows, other than to test the Windows versions of programs I publish.

tlfisher, 2016-09-25:
Paul, I think perhaps many Windows users who would like to use teensy-loader-cli are stymied because we cannot compile the .c file.  I cannot find the required include files even after installing Visual Studio 2015.  Could you possibly post the loader in the form we could use on the command line?  Sorry to bother you, but I don't know another way to obtain this file.  Thank you.

ctadlock, 2024-03-10:
Get the compiled from PlatformIO.

https://forum.pjrc.com/index.php?threads/teensy_loader_cli-executable-for-windows.63176/page-2#post-340526