PaulStoffregen/teensy_loader_cli #24: Win32 Flashing Teensy 2.0 / Teensy 3.2 / Teensy LC

rjonkman, 2016-05-17:
I'm having issues getting the the loader to work reliably for all these various controllers on Win 32. The following changes implemented by josh3io seem to work, but they look like a bit of a hack and result in very slow uploading:

https://github.com/josh3io/teensy_cli_loader_win32

Paul, the Teensy.exe app works flawlessly for all the these chips.  Would it be possible for you to post the  write_usb_device() and teensy_write() methods that you use in the Teensy.exe program.  If you were willing to, I'd update the teensy_loader_cli methods and test them to resolve the current issues on Windows.

Regards,
Rob

rjonkman, 2016-05-17:
I think I found a good solution by modifying write_usb_device()

...
       int waited = 0;
	while (!GetOverlappedResult(h, &ov, &n, FALSE) || n <= 0) {
		Sleep(1);
		waited++;
		if (waited > timeout) return 0;
	}

	return 1;

rjonkman, 2016-05-17:
...Doesn't work... It times out...

rjonkman, 2016-05-17:
This seems to be working...
```
int teensy_write(void *buf, int len, double timeout)
{

	if (!win32_teensy_handle) return 0;

	int waited = 0;

	while (waited < timeout * 1000) {
		
		if (write_usb_device(win32_teensy_handle, buf, len, (int)(timeout * 1000.0))) return 1;

		Sleep(10);
		waited += 10;

	}
	return 0;
}

int write_usb_device(HANDLE h, void *buf, int len, int timeout) {
	static HANDLE event = NULL;
	unsigned char tmpbuf[1089];
	OVERLAPPED ov;
	DWORD n, r;
    
	if (len > sizeof(tmpbuf)-1) return 0;
	
	if (event == NULL) {
		event = CreateEvent(NULL, TRUE, TRUE, NULL);
		if (!event) return 0;
	}

	ResetEvent(&event);
	memset(&ov, 0, sizeof(ov));
	ov.hEvent = event;
	tmpbuf[0] = 0;
	memcpy(tmpbuf + 1, buf, len);
	bool res;

	res = WriteFile(h, tmpbuf, len + 1, NULL, &ov);

	if (!res) {
		if (GetLastError() != ERROR_IO_PENDING) return 0;
	}
	
	res = GetOverlappedResult(h, &ov, &n, TRUE);

	if (!res) return 0;
	
	return 1;
}

```

PaulStoffregen, 2016-11-09:
Fixed  https://github.com/PaulStoffregen/teensy_loader_cli/commit/0416adf3d250bf1d82913be514aaa51cb1d5f098

closed 2016-11-09 by PaulStoffregen