PaulStoffregen/teensy_loader_cli #28: Support for 3.5 / 3.6

ivankravets, 2016-09-09:
Thanks.

blackketter, 2016-09-27:
I believe this was fixed in https://github.com/PaulStoffregen/teensy_loader_cli/pull/27

ivankravets, 2016-09-27:
Could you help us with binaries? I'll update all packages in https://bintray.com/platformio/dl-packages/tool-teensy/

Thanks!

PaulStoffregen, 2016-09-27:
I can't do anything until next week.

blackketter, 2016-10-05:
I'd be happy to help out here, how/where are the binaries built?

chrisJohn404, 2016-10-26:
Not sure if people are struggling/trying to figure out how to write code for the Teensy 3.5 but I'm just getting started with the Teensy and this is also the first arm based MCU that I have used.  I used to do a good bit of embedded things, but it has been a while so I definitely fall closer to the NOOB category.  That being said, any and all ARM development that I have tried to do in the past has been fairly hard to understand and the docs for the 3.5/3.6 seem to be all over the place.  Its hard to tell what docs apply to only the AVR based Teensy's vs the ARM based ones.  I'm sure all of the information exists, I am just having a hard time finding it.  I also have no clue where a post like this should go but wanted to speak up in case someone else ends up in this repo like I did and is fairly lost.   

I believe I came across this .git repo by trying to follow the [Teensy Tutorial 1: Software Setup](https://www.pjrc.com/teensy/tutorial.html).  After installing the Teensy Loader application and not finding .hex files for the Teensy 3.5 and 3.6, I stumbled upon this git repo & the link to PlatformIO.  I also failed to properly install the Teensyduino installer due to not knowing where the Arduino files got extracted on my machine.  I assume this repo is tied pretty closely to the teensy loader GUI application but I'm not entirely sure how.

I was able to get things working pretty easily with the PlatformIO IDE linked by @ivankravets and compile/upload the two example projects advertised for the Teensy in the PlatformIO IDE:
[PlatformIO Examples/Teensy](https://github.com/platformio/platformio-examples/tree/develop/teensy).

I am still trying to work through everything I had to install to get things working, but I believe I did something like the following:
1. Install the [Teensy Loader App (Version 1.21)](http://www.pjrc.com/teensy/loader.html)

2. I didn't already have Github's Atom text editor installed so I ran the windows installer advertised by PlatformIO's website.

3. When Atom first launched it installed the PlatformIO IDE extension and I was able to initialize both of the Teensy examples by going to the menu link: PlatformIO -> Project Examples, individually selecting both teensy projects, and pressing the "prepare" button.  On my Windows 10 machine they got initialized in a temporary directory.

4. After that I believe I was instructed by Atom/PlatformIO to install Clang which sent me to this page:
[PlatformIO IDE for Atom](http://docs.platformio.org/en/stable/ide/atom.html#installation).  I already have Python installed and a variety of MSBuild things due to working on native-code for Node.js, but make sure you add LLVM to the system PATH variable.

5. I was then able to build the two examples.  FYI, you can press "F8" to re-display the build-output.

6. The two projects automatically built for the other Teensy boards (<3.5/3.6) so I had to add the teensy35 as a target.  I believe this can be done in the platformio.ini file in the example directory as well as through the "PlatformIO -> Initialize or Update PlatformIO Project" menu-link.  Wasn't sure what would happen, but according to PlatformIO's [Platform Teensy](http://docs.platformio.org/en/stable/platforms/teensy.html?highlight=Teensy) Docs page, it looked like 3.5 and 3.6 were supported.  

I then got rid of the other Teensy board targets to speed up the build process (directly edited the .ini file).

7. I then built the project again: "Alt+Ctrl+B". 

8. After building, I uploaded the project to the board: "Alt+Ctrl+U".  When first doing this I didn't realize that I still had the Teensy Loader App was still running and PlatformIO was actually using the program/service made available by the loader to load the .hex file onto the board.  After uploading, it also took me a second to realize that I needed to carefully follow the last sentence in the "Download LED Blink" step in the [Teensy Tutorial 1: Software Setup](https://www.pjrc.com/teensy/tutorial.html) tutorial saying:
> Just click the "Program" button to write this new code to your Teensy, and then click the "Reboot" button to cause it to run. Your Teensy should be blinking rapidly. You have verified everything works and you're ready to use it with Arduino.
Running the ChatServer example required me to press the reboot button to get any text-output changes to show up in the Arduino's serial monitor tool.

9. After getting completing the previous steps, I was able to copy the projects into a new location, change the folder names and incorporate the Arduino Blink example to create a test project "teensy-fast-blink" which gave me a firmware.hex and .elf files that I re-named to "blink_fast_Teensy35".  I believe people should be able to download & extract the .zip file and then add the folder to Atom by "Alt+Ctrl+O" or the file -> Add Project Folder menulink.

Below is the .hex file (in the .zip file) for the blink_fast demo that was referenced by the initial tutorial I tried to follow as well as the PlatformIO project.  Hopefully the [blink_both.zip](https://www.pjrc.com/teensy/blink_both.zip) file will eventually get updated to have the Teensy 3.5 & 3.6 pre-built code.
[teensy-fast-blink.zip](https://github.com/PaulStoffregen/teensy_loader_cli/files/552028/teensy-fast-blink.zip)
[elf-and-hex-files.zip](https://github.com/PaulStoffregen/teensy_loader_cli/files/552031/elf-and-hex-files.zip)

Any chance someone could answer a few questions I came up with during this process?
[... cut ...]
Links to .pdf's, tutorials, other things to read are totally welcome.  I'm probably missing some sort of fundamental piece of information and just don't know what to google/where to look & just need to be pushed in the right direction.

Thanks @PaulStoffregen for all of your hard work and creating/selling the Teensy boards as well as everyone else involved in the project.  I look forward to learning more about these processors/ARM in general and hopefully being able to give back to the community one day.  Sorry if this post should have been posted somewhere else and isn't related to this github issue.

ivankravets, 2016-10-28:
Dear @chrisJohn404,

> 1. In the README.rst file, there should have been a "platformio" command added to the system PATH variable.

Where do you see that? If you use PlatformIO IDE, you don't need to play with PATH or etc. Just open PlatformIO IDE Terminal where PlatfomIO CLI is built-in here.

> 2 & 3 What is the connection between PlatformIO

PlaformIO is independent, free and open source ecosystem with build system without any dependencies to OS or Arduino IDE. We have separate [Teensy development platform](https://github.com/platformio/platform-teensy) that is based on PlatformIO Build System API.

chrisJohn404, 2016-10-29:
> Where do you see that?

I saw this in the README.rst files in the two downloadable PlatformIO examples.
1. [teensy/teensy-hid-usb-mouse/README.rst](https://github.com/platformio/platformio-examples/tree/develop/teensy/teensy-hid-usb-mouse)
2. [teensy/teensy-internal-libs/README.rst](https://github.com/platformio/platformio-examples/tree/develop/teensy/teensy-internal-libs)

Thanks for the help!  After reading your second response and following your link I realized that I just need to install the [PlatformIO CLI](http://docs.platformio.org/en/stable/installation.html#python-package-manager).   I already had python installed and pip too so the installation was pretty easy.  I just ran the command:
`> pip install -U platformio`

ivankravets, 2016-11-02:
@chrisJohn404 do you need any help from @PlatformIO side?

PaulStoffregen, 2016-11-09:
I'm trying to understand what this issue is about.  Teensy 3.5 and 3.6 support has been in the code for months.

Is this a request for a mac binary?

ivankravets, 2016-11-09:
@blackketter I don't have these boards. Please build manually https://github.com/PaulStoffregen/teensy_loader_cli and check if it works. If yes, please attach here macOS binary. I'll put this binary to PIO Package Repository.

Thanks!

PaulStoffregen, 2016-11-09:
I've added Windows and Macintosh binaries on the main page.

http://www.pjrc.com/teensy/loader_cli.html

Windows, Linux and Macintosh now work for all boards.

I'm going to close this issue.  If there's still any problem, please open a new issue.  Please be specific and focus on 1 problem per github issue.  I have very limited time to maintain this command line version, so please help me with specific issues.

closed 2016-11-09 by PaulStoffregen

ivankravets, 2016-11-14:
@PaulStoffregen I've just release new Teensy development version https://github.com/platformio/platform-teensy/releases/tag/v1.2.2

Thank you for your binaries. I also compiled for Linux 32/64 and ARMv6 (RaspberryPi and etc)