PaulStoffregen/teensy_loader_cli #30: usb.h is not found

kamikaze, 2016-11-05:
Looks like there is a build error with new libusb:

```
cc -O2 -Wall -s -DUSE_LIBUSB -o teensy_loader_cli teensy_loader_cli.c -lusb
teensy_loader_cli.c:215:17 fatal error: usb.h: No such file or directory
```

I have libusb-1.0.21 installed on Gentoo

neonsoftware, 2016-11-08:
Hi @kamikaze . 

to my knowledge teensy_loader_cli used to be based on libusb version 0.1, which has usb.h. It used to be written in the comments in the code but can't remember where and when. Please anyone correct me if I am wrong.

Version 1.0 of libusb, which you have installed, has a new and different API from 0.1. (0.1 and 1.0 are two _'era_' of the libusb library, and both are still separately used and packaged, long story ) 

Anyways one versatile and easy solution that allows you to use both the API of 0.1 and 1.0 (in case you will need it) is to use the package **libusb-compat**, look for it on your package manager, debian has it, hope gentoo too.

[ fyi what internally that package does is providing 1.0 as it is, and also providing headers and wrapper functions that provide the 0.1 API, where the internal implementation is actually done using 1.0 ] 

You could uninstall your libusb-1.0.21 before installing **libusb-compat**

Hope it works :)

PaulStoffregen, 2016-11-08:
Can anyone suggest specific text to document this in the comments in the code?  Ideally that text would have the exact package names or info needed for various Linux distros which don't provide this header by default.

neonsoftware, 2016-11-09:
I'll attempt to try-out cloning&compilation on the various distributions, clean install, and note down the steps.  
By now on Debian,Ubuntu and ArchLinux the package _libusb-compat_ is available in the package manager and gets the compilation done. Next I will check Red-Hat, Fedora, and Gentoo.  

I will post updates and info here. Any help or request of distro to try is welcome.

kamikaze, 2016-11-09:
why not to use newer libusb?

PaulStoffregen, 2016-11-09:
> why not to use newer libusb?

Do you want to update the code and test throughly?  I know I have much better things to do.

neonsoftware, 2016-11-11:
Hi @PaulStoffregen, 

commit   https://github.com/neonsoftware/teensy_loader_cli/commit/e9683457d99dc21b7fe7617e2da1140c28e9ba3f adds text to document libusb0.1 installation on various linux distros and freebsd.
Please let me know if anything can be added or modified. If it is interesting to you then I'll make a pull request

neonsoftware, 2016-11-11:
I have tested it against the Linux distros by setting up a CI job that runs on Travis.
I have added the use of the 'latest' tag where available, so it will always check against the latest distribution release.
It is in https://github.com/neonsoftware/teensy_loader_cli/commit/bf488b3bbb15c0f7921c45f56fb4f52ce79d08bb if you are interested, but otherwise I will keep the fork updated and let you know when newer releases break.

Regarding freebsd, I tested by creating a virtual machine with version 10.3-release, if other versions must be checked, let me know.

PaulStoffregen, 2016-11-11:
> If it is interesting to you then I'll make a pull request

Yes, please do, for https://github.com/neonsoftware/teensy_loader_cli/commit/e9683457d99dc21b7fe7617e2da1140c28e9ba3f

neonsoftware, 2016-11-11:
pull https://github.com/PaulStoffregen/teensy_loader_cli/pull/31

neonsoftware, 2016-11-11:
👍  and thanks for all the good work you put in the project

closed 2022-09-09 by PaulStoffregen