PaulStoffregen/teensy_loader_cli #37: Soft reboot is not implemented for OSX

chappy1978, 2017-04-11:
Hey Paul,
 Is there any plan to implement soft reboot on OSX in the CLI? If not, why? When I load a .hex with the app it works fine. What would it take to make it work?
  Jeff

PaulStoffregen, 2017-04-11:
No plans.  In fact, I'm specifically planning to not do anything more on this CLI tool for at least the rest of 2017.

> What would it take to make it work?

Someone willing to do all the work, and submit a low-risk pull request.  Edits to other parts of the code which may break things for other platforms almost certainly will not be accepted, so it needs to not only work, but be done in a clean way that doesn't risk breaking other platforms.  Several contributions I receive are written badly and break things for other users, so I am weary of accepting such things.

chappy1978, 2017-04-11:
> No plans. In fact, I'm specifically planning to not do anything more on this CLI tool for at least the rest of 2017.
>
> What would it take to make it work?
>
> Someone willing to do all the work, and [...]
 Sadly, Im sure thats just over my head. Thanks for the quick answer.