PaulStoffregen/teensy_loader_cli #40: put HalfKay mode without pressing the button

varadig, 2018-01-26:
There is a way to put the teensy(3.6) to HalfKay mode, with CLI, without pressing the reset button.
I can u[load the HEX file to the board with CLI, but just if I press the reset button:

```
>teensy_loader_cli.exe --mcu=mk66fx1m0 -v -w teensy_firmware_update.ino.hex
Teensy Loader, Command Line, Version 2.1
Read "teensy_firmware_update.ino.hex": 12672 bytes, 1.2% usage
Waiting for Teensy device...
 (hint: press the reset button)
```

Im use windows, and with the teensy.exe I can upload program to teensy with automatic mode, but I need to use the CLI version of loader.

varadig, 2018-01-26:
I try to use -r arg to hard reboot. but I got like this:
`teensy_loader_cli.exe -r -v --mcu=mk66fx1m0 firmware.hex`
But I got:

```
Teensy Loader, Command Line, Version 2.1
firmware.hex": 12672 bytes, 1.2% usage
**Unable to find rebootor**
```

vincenet, 2018-02-06:
> Unable to find rebootor

>-r : Use hard reboot if device not online. Perform a hard reset using a second Teensy running this rebootor code, with pin C7 connected to the reset pin on your main Teensy. While this requires using a second board, it allows a Makefile to fully automate reprogramming your Teensy. No manual button press is required! 
[https://www.pjrc.com/teensy/loader_cli.html](url)

pryankster, 2018-06-06:
This has been around a while ... but is there any way to get halfKay running ?  I have a Teensy3.1 (Might be a 3.2 -- I've had it for a while, and it's soldered to an audio shield, so I can't see the bottom of the board :-)

When I connect it to my raspberry PI, and run "lsusb":
```# lsusb -d 0x16c0:
Bus 001 Device 033: ID 16c0:0489 Van Ooijen Technische Informatica```

I don't see `16c0:0478` unless I press the button:
```# lsusb -d 0x16c0:
Bus 001 Device 034: ID 16c0:0478 Van Ooijen Technische Informatica Teensy Halfkay Bootloader
```
I want to embed the teensy in a project that I'm building with the Raspberry PI, and I want the PI to be able to update the firmware on the Teensy in a "hands-off" manner, if needed.

The teensy is configured with USB as "Serial + MIDI".  I have tried the "-s" soft reboot option.  The RaspberryPI is headless, so I can't use the "auto" mode of the GUI programmer 

Any suggestions would be greatly appreciated!

-- pryankster

vincenet, 2018-06-06:
I trig a signal from Linux board to make the pressing for me.

pryankster, 2018-06-06:
Right now that's doable -- but the eventual product will be using the DPI interface for display which eats up all of the RPI's GPIOs (which is why I'm using a Teensy in the first place do manage I/O over USB to the PI.

vincenet, 2018-06-07:
And using a second teensy with rebootor code ?

pryankster, 2018-06-07:
I don't want to add a whole third processor just to reset the second processor!

vincenet, 2018-06-13:
With Dragino board based on openWrt, I get directly the teensy at start-up and lsusb returns : 
`Bus 001 Device 004: ID 16c0:0483 Van Ooijen Technische Informatica Teensyduino Serial`
uname -r returns 3.3.8
If you solves this first point you can may be accept to use the software reboot (with may be little risk to block the system compared to real hardware reboot).

siteswapjuggler, 2020-06-19:
Hello @PaulStoffregen any news on this. I would be happy to help but I'd greatly appreciate to get some starting point to work with.

The desktop version of TeensyLoader seems able to reboot Teensys in halfkay without problem, do you have any source code to share that we could try to implement in the cli version ?

In parralel I'm working to make a selection by SerialNumber so we can select easily which card to work with and update multiple cards in the command line.

I've found this article on the PJRC website : https://www.pjrc.com/teensy/jump_to_bootloader.html the triggering ideas mention to send a  "custom endpoint 0 message" but I didn't find anything close to this on my first research.

paqwalsh, 2020-11-16:
I'm able to force a reset from which the loader is able to do it's job from software alright. I'm using the less than ideal method of calling a function which exists in the core_pins library of the Teensy core as follows: 
`_reboot_Teensyduino_();`
This works fine if you set up your serial reading functions to interpret some custom input of your choosing to then run it when detected so that you can call a reset from another device and then you can initiate the firmware update process. 

What I would much prefer to do is figure out where in the core libraries this function is called and how it is called, i.e. what sequence of bytes required over UBS to get the core library to run it (as is done with the teensy's bootloader programs). This would save me having to check for it on the incomming buffer using Serial.read() as the core library would handle it and I wouldn't have to worry about what state my own software is in when I call it, i.e. if I'm stuck in a loop that not reading the incomming serial stream! 

Hope this helps and if you find out way to reset it from the core libraries please share, or if @PaulStoffregen could comment, it'd be very grateful, thanks.

paqwa, 2022-06-17:
I’m pretty sure you can enter programming mode by simply disconnecting from Teensy  device from PC via software and then reconnecting momentarily using a specific baud rate. I’ve done it before with Teensy 3.6 when developing my own app but not at PC this moment so can’t recall exact baud rate to use! You can find it in the source code for Teensy Cli.

PaulStoffregen, 2022-06-17:
> there is no automatic reboot on OSX

Look for the "teensy_reboot" command line program.  It's inside the Teensyduino app bundle, so if using Finder, control-click Teensyduino and click "Show package contents" to look inside.

No plans to build all this into teensy_loader_cli.

closed 2022-06-17 by PaulStoffregen

steeley, 2022-06-18:
> I’m pretty sure you can enter programming mode by simply disconnecting from Teensy device from PC via software and then reconnecting momentarily using a specific baud rate. I’ve done it before with [...]

Thanks Paqwa  - Played with this but never got it working ... We have now decided to move away from Teensy to ESP32   as we can do auto/remote updates much more easily. Also OTA via wireless if needed.

pictographer, 2022-06-19:
For what it's worth, the way I do OTA programming of the Teensy is this:
- Teensy 3.2 or Teensy LC connected by USB serial to a GL-iNet Mango Wifi Router running OpenWrt Linux
- Use `scp` to copy the hex file from my desktop system to the router
- The router runs `usbreset "16c0:0478"`
- The router sends a "reboot" command to the Teensy over USB Serial.
- The firmware on the Teensy calls `_reboot_Teensyduino_()` when it receives the "reboot" command
- The router runs `teensy_loader_cli --mcu=mk20dx256 -v -s cove1.ino.hex`
- If the Teensy doesn't reboot into program mode, the router can power cycle the Teensy
 
 I don't know why `usbreset` is needed, but after I added it, the solution above became reliable.

The script on the router to send command to the Teensy uses `stty -F /dev/ttyACM0 -echo -icrnl -onlcr`. Getting output from the Teensy on OpenWrt is a bit of a pain. The full script is this:

```sh
#!/bin/ash

do_stty() {
    stty -F /dev/ttyACM0 -echo -icrnl -onlcr
}

do_reset() {
    usbreset 16c0:0487 > /dev/null
    sleep 1
}

do_stty || (do_reset; do_stty)
exec 3< /dev/ttyACM0
echo "$@" > /dev/ttyACM0
read -u 3 -t 2 result && echo "${result}" || echo "read timedout for ttyACM0."
exec 3<&-
```