PaulStoffregen/teensy_loader_cli #46: Teensy_loader_cli not releasing comm port after reboot.

duskatnine, 2019-07-12:
I am running teensy_loader_cli on a raspberry pi.

If I check my USB Ports before and after a successful Teensy Loader execution I see the teensy loader has not released the USB Serial Port. Any thoughts?

/* this is my command to reprgram the teensy 3.2 with a soft reboot (-s) */
sudo ./bootloader/teensyLoader/teensy_loader_cli --mcu=mk20dx256 -s -v ./bootloader/sw_version/BlinkSlow.hex

/* Here is the terminal */
> Soft reboot performed
> Waiting for Teensy device...
> (hint: press the reset button)
> Found HalfKay Bootloader
> Read "./bootloader/sw_version/BlinkSlow.hex": 67996 bytes, 25.9% usage
> Programming...................................................................
> Booting

/* if I read my USB ports before the successful update ( ls /dev/tty* ) */
I see /dev/ttyACM0 before ( which is what my teensy is connected to ) but not after

PaulStoffregen, 2019-07-12:
Not a bug.  Teensy uses HID protocol for uploading code, not serial.

If you want to ask more questions, please do so on the forum.  This issue tracker is *only* for reporting defects in the software.  This is not a defect.

closed 2019-07-12 by PaulStoffregen

duskatnine, 2019-07-12:
Hi Paul!
Apologies, will troubleshoot via the forum.

Thank your for your repo! Pleasure to use.  Cheers