PaulStoffregen/teensy_loader_cli #47: Add support for Teensy 4.0

blackketter, 2019-08-12:
Not sure what's needed, but I'm happy to pitch in and help.

branan, 2019-08-19:
I was just looking in to this today. *Assuming* that the protocol hasn't changed, the changes are pretty minimal:

* Update the max firmware size to 2MB at https://github.com/PaulStoffregen/teensy_loader_cli/blob/master/teensy_loader_cli.c#L842
* Add appropriate entries for the Teensy4 to the MCU list at https://github.com/PaulStoffregen/teensy_loader_cli/blob/master/teensy_loader_cli.c#L1051-L1078
  * While you're in there, maybe fix the bug where the friendly names for the Teensy2s only show up if USB support is enabled

I think the only open question for this is what the block size is for the flash. https://github.com/PaulStoffregen/cores/blob/master/teensy4/bootdata.c#L191 seems to indicate it's 64KiB, but I'm unsure if the bootloader expects 64K blobs or if it takes them in smaller chunks over the USB.

It would probably take someone an hour or two to trial-and-error whether the protocol is the same and what the necessary block size is.

d-huck, 2019-08-23:
I was able to get it working with Teensy 4.0 after a rather big headache. 

It looks like the Teensy 4.0 hex files have this fun little line added to the beginning:

```
:0200000460009A
```

According to the [Intel Hex Format](https://en.wikipedia.org/wiki/Intel_HEX#Record_types), this sets the upper 16 bits of the 32-bit memory address. It uses the value 0x6000 to set this value. This then sets the extended_addr to 0x60000000, which suggests there is almost 1.5GB of memory available on the Teensy 4.0!

I changed the MAX_MEMORY_SIZE to 0x200000 as well as edited [this line of code](https://github.com/PaulStoffregen/teensy_loader_cli/blob/eb61aa1a482c2ad98916229e02bb8caae2464f48/teensy_loader_cli.c#L1051-L1078) to look like the following:

```c++
static const struct {
	const char *name;
	int code_size;
	int block_size;
} MCUs[] = {
	{"at90usb162",   15872,   128},
	{"atmega32u4",   32256,   128},
	{"at90usb646",   64512,   256},
	{"at90usb1286", 130048,   256},
#if defined(USE_LIBUSB) || defined(USE_APPLE_IOKIT) || defined(USE_WIN32)
	{"mkl26z64",     63488,   512},
	{"mk20dx128",   131072,  1024},
	{"mk20dx256",   262144,  1024},
	{"mk66fx1m0",  1048576,  1024},
	{"mk64fx512",   524288,  1024},
    {"imxrt1062",  2097152,  1024},

	// Add duplicates that match friendly Teensy Names
	// Match board names in boards.txt
	{"TEENSY2",   32256,   128},
	{"TEENSY2PP", 130048,   256},
	{"TEENSYLC",     63488,   512},
	{"TEENSY30",   131072,  1024},
	{"TEENSY31",   262144,  1024},
	{"TEENSY35",   524288,  1024},
	{"TEENSY36",  1048576,  1024},
    {"TEENSY40",  2097152,  1024},
#endif
	{NULL, 0, 0},
};
```
In order to get around the problem I mention above, I also changed [this line of code](https://github.com/PaulStoffregen/teensy_loader_cli/blob/eb61aa1a482c2ad98916229e02bb8caae2464f48/teensy_loader_cli.c#L931-L939) to ignore any commands to use an extended address above the MAX_MEMORY_SIZE.

```c++
if (code == 4 && len == 2) {
	if (!sscanf(ptr, "%04x", &i)) return 1;
	ptr += 4;
	sum += ((i >> 8) & 255) + (i & 255);
		if (!sscanf(ptr, "%02x", &cksum)) return 1;
	if (((sum & 255) + (cksum & 255)) & 255) return 1;
        if ((i << 16) > MAX_MEMORY_SIZE) return 1;
	extended_addr = i << 16;
	printf("ext addr = %08X\n", extended_addr);
}
```

This solution is admittedly hacky, but it has worked with several different hex files I have.

luni64, 2019-09-03:
> It looks like the Teensy 4.0 hex files have this fun little line added to the beginning:
> :0200000460009A

Makes sense if you look at the linker skript: 

imxrt1062.ld
[image]
Looks like this is some virtual address the flash is mapped to. Just wondering if ignoring this adress is the right way to do it but if it works...

Maybe Paul can shed some light into that?

mpasternacki, 2019-09-09:
If somebody feels like experimenting, I just published a first version of loader CLI reimplementation in Rust that should work with 4.0 (but haven't tested it with a 4.0 yet – I still need to get some binary working; tested with an LC, though, and tested booting from HalfKay into user mode with 3.6 and 4.0): https://crates.io/crates/teensy-loader

Caveats: libusb only (Linux/FreeBSD), accepts ELF file as an input (because I was tired of having to convert to ihex); patches for supporting other platforms / input formats are welcome, though.

PaulStoffregen, 2019-10-08:
Done  https://github.com/PaulStoffregen/teensy_loader_cli/commit/900a2bf56ed8f0627127dadfde8742e9810eff63

closed 2019-10-08 by PaulStoffregen