PaulStoffregen/teensy_loader_cli #54: Release with Teensy 4.0 support

Walther, 2020-03-15:
Thank you for this great tool!

Teensy 4.0 support has been added in https://github.com/PaulStoffregen/teensy_loader_cli/commit/900a2bf56ed8f0627127dadfde8742e9810eff63

Installing `teensy_loader_cli` from Homebrew gets you the version
```
teensy_loader_cli: stable 2.1 (bottled), HEAD
Command-line integration for Teensy USB development boards
https://www.pjrc.com/teensy/loader_cli.html
/usr/local/Cellar/teensy_loader_cli/2.1_1 (4 files, 36.4KB) *
  Poured from bottle on 2020-03-15 at 20:00:26
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/teensy_loader_cli.rb
```

List of supported devices:
```
teensy_loader_cli --list-mcus
Supported MCUs are:
 - at90usb162
 - atmega32u4
 - at90usb646
 - at90usb1286
 - mkl26z64
 - mk20dx128
 - mk20dx256
 - mk66fx1m0
 - mk64fx512
```

Attempting to run for a Teensy 4.0 as per instructions:

```
teensy_loader_cli --mcu=imxrt1062 out/led.hex
Unknown MCU type "imxrt1062"
Supported MCUs are:
 - at90usb162
 - atmega32u4
 - at90usb646
 - at90usb1286
 - mkl26z64
 - mk20dx128
 - mk20dx256
 - mk66fx1m0
 - mk64fx512
```

It would be nice to get a release with the Teensy 4.0 support built in ❤️

palani-thangaraj, 2020-04-01:
So is it or is it not compatible with 4.0 yet?

Walther, 2020-04-01:
Supposedly, the code in this repository works, if you compile it from source.
The version that has been released and can be installed via Homebrew is older, and does not include that support yet.

palani-thangaraj, 2020-04-01:
What is currently the easiest way to CLI upload new code. I essentially just need to send code wirelessly to a raspbery pi and then from there it automatically uploads the code to the teensy 4.0.  I haven't been able to get the teensy CLI to work, some people suggested sending the built hex file or using platformIO but I don't know much about either

Ben-Lichtman, 2021-09-06:
Just wanted to bump this, since it has been >1.5 years and there has still not been a versioned release of this. Therefore all of the package manager still hold versions of this tool which don't support teensy 4.0

uqs, 2021-12-05:
+1, most distros are stuck at the (by now ancient) v2.1 w/o support for 4.0 and 4.1

PaulStoffregen, 2021-12-05:
done
https://github.com/PaulStoffregen/teensy_loader_cli/releases/tag/2.2