PaulStoffregen/teensy_loader_cli #55: teensy_loader_cli.c:373:10: fatal error: ddk/hidsdi.h: No such file or directory

thjwong, 2020-06-07:
I tried to compile the package in Windows 10 (64-bits) under MSYS2 and encountered the following errors:

1. `printenv` gave me "Windows_NT" instead of "WINDOWS", so I had to change that.
2. then there is an error message for ddk/hidsdi.h

> teensy_loader_cli.c:373:10: fatal error: ddk/hidsdi.h: No such file or directory
  373 | #include <ddk/hidsdi.h>

According to https://packages.msys2.org/package/mingw-w64-x86_64-headers
hidsdi.h is not under ddk for MSYS2, so I went to line 373 and changed that too.

3. then there are other error messages regarding ddk/hidclass.h

> C:/msys64/mingw64/x86_64-w64-mingw32/include/ddk/hidclass.h:132:3: error: unknown type name 'PINTERFACE_REFERENCE'
  132 |   PINTERFACE_REFERENCE  InterfaceReference;
      |   ^~~~~~~~~~~~~~~~~~~~
C:/msys64/mingw64/x86_64-w64-mingw32/include/ddk/hidclass.h:133:3: error: unknown type name 'PINTERFACE_DEREFERENCE'
  133 |   PINTERFACE_DEREFERENCE  InterfaceDereference;
      |   ^~~~~~~~~~~~~~~~~~~~~~
C:/msys64/mingw64/x86_64-w64-mingw32/include/ddk/hidclass.h:147:3: error: unknown type name 'PINTERFACE_REFERENCE'
  147 |   PINTERFACE_REFERENCE  InterfaceReference;
      |   ^~~~~~~~~~~~~~~~~~~~
C:/msys64/mingw64/x86_64-w64-mingw32/include/ddk/hidclass.h:148:3: error: unknown type name 'PINTERFACE_DEREFERENCE'
  148 |   PINTERFACE_DEREFERENCE  InterfaceDereference;

Please give me a pointer to compile it under msys2 if it is possible.
Thanks!

fauxpark, 2020-08-20:
@thjwong I've added teensy_loader_cli as an MSYS2 package :)
https://packages.msys2.org/base/mingw-w64-teensy-loader-cli

ser0ja, 2020-11-06:
Please update a package for Teensy 4.0
```
# teensy_loader_cli --mcu=imxrt1062 -w blink_slow_Teensy40.hex
Supported MCUs are:
 - at90usb162
 - atmega32u4
 - at90usb646
 - at90usb1286
 - mkl26z64
 - mk20dx128
 - mk20dx256
 - mk66fx1m0
 - mk64fx512
Unknown MCU type "imxrt1062"
```

PaulStoffregen, 2020-11-07:
@ser0ja - Download the latest code.  You must have an old copy.  It was updated for Teensy 4.0 over 1 year ago!  https://github.com/PaulStoffregen/teensy_loader_cli/commit/900a2bf56ed8f0627127dadfde8742e9810eff63

fauxpark, 2020-11-07:
@PaulStoffregen can we get a new release so packagers can update?

ser0ja, 2020-11-07:
@PaulStoffregen
https://packages.msys2.org/base/mingw-w64-teensy-loader-cli

[image]

[image]

hmaarrfk, 2021-04-10:
I believe I tried to address a similar issue in:
https://github.com/PaulStoffregen/teensy_loader_cli/pull/59

You can check if the patch works for you.