PaulStoffregen/teensy_loader_cli #56: Is there a way to specify which teensy is programmed via its serial number?

hmaarrfk, 2020-07-17:
For use cases where we have multiple teensies connected, is it possible to specify which one we program via their serial number? usb port?

takurx, 2021-02-11:
Hi @hmaarrfk,
I also want to write to multiple teensy (same version, 3 teensy 4.1s).
When it open Teensy, it specify vid and pid.
https://github.com/PaulStoffregen/teensy_loader_cli/blob/master/teensy_loader_cli.c#L669
Teensy have same vid and pid.
I think that new function option need to specify usb port.

hmaarrfk, 2021-02-11:
You can try 

https://github.com/PaulStoffregen/teensy_loader_cli/pull/57

Only works on linux.

I did the bad thing on releasing the feature on conda forge as well

takurx, 2021-02-12:
Thank you. That's helpful for me!! I will try it.

FiratSusan, 2021-05-27:
Nevermind, your are just comparing the strings according to the length of the given serial_number (so the trailing 5 is ignored).

FiratSusan, 2021-05-31:
In the run shown above no.

hmaarrfk, 2021-05-31:
are you using a standard intel based computer? I've been using this for quite some time and it seems to be "working" but maybe there is a bug....