PaulStoffregen/teensy_loader_cli #60: Possible approach for macOS: Error opening HID Manager

jonbakke, 2020-07-29:
I encountered the "error opening HID Manager" problem and tried to dig in deep enough to understand some of it. My -- extremely inexpert -- conclusion is that this is caused by a security measure that requires a sandboxed environment before IOHIDManagerOpen will return a meaningful value, I suppose to prevent using it to probe the surface of everything attached to the machine via USB. I'm now comfortable ignoring this problem and have changed the code I use accordingly, yet not confident enough to issue a pull request. If discussion leads toward including similar code, I'd be happy to polish and make a PR; otherwise, this issue might act as a signpost to others.

I also expanded the error message from what I found in the IOkit headers. If something based on that would be valuable, let me know and I'll clean up its crude style/placement to fit better with the existing code.

My fork is here: https://github.com/jonbakke/teensy_loader_cli/blob/master/teensy_loader_cli.c (lines 622 et seq). 

Regardless -- loving my Teensy. Thanks, Paul.

chrismoos, 2020-08-01:
Thanks @jonbakke I had the same issue and it fixed it for me. Any chance you can send a PR here with your fix?

noisymime, 2020-10-21:
Another +1 here for this. Fix works well for me, thanks!

avivbeeri, 2022-12-29:
This is still an issue and @jonbakke's fork worked for me, unlike everything else I've tried.