PaulStoffregen/teensy_loader_cli #69: Makefile:59: recipe for target 'clean' failed

daanvV, 2021-06-10:
Using Windows 10, installed MinGW

I get the following error when doing this command: 
`(base) PS C:\XXXX\teensy_loader_cli-master> mingw32-make`

Error: 
`rm -f teensy_loader_cli teensy_loader_cli.exe*
process_begin: CreateProcess(NULL, rm -f teensy_loader_cli teensy_loader_cli.exe*, ...) failed.
make (e=2): The system cannot find the file specified.
Makefile:59: recipe for target 'clean' failed
mingw32-make: *** [clean] Error 2`

When I then do: 
`(base) PS C:\XXXteensy_loader_cli-master> teensy-loader-cli`

I get: 
```
teensy-loader-cli : The term 'teensy-loader-cli' is not recognized as the name of a cmdlet, function, script file, or
operable program. Check the spelling of the name, or if a path was included, verify that the path is correct and try
again.
At line:1 char:1
+ teensy-loader-cli
+ ~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (teensy-loader-cli:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

```
Any help would be much appreciated!

PaulStoffregen, 2022-06-23:
Maybe you're missing the "rm" command.  I believe in Windows a similar command is "del", though it might use different syntax than "-f".  Probably easier to install a rm command somehow.

ZanzyTHEbar, 2022-06-23:
yeah, i figured that out - so i tried on ubuntu but i can't seem to install MXE - MXE build fails and the binanry install says the certificate is invalid - when i try to build without MXE - of course it can't find `windows.h`, which makes sense. So, i am at a loss of _exactly_ how you compile your software. You claim it "just works" - but so far that is anything but true. Any guidance?

ZanzyTHEbar, 2022-06-24:
[here is the source for the claim](https://github.com/PaulStoffregen/teensy_loader_cli/issues/23)

Which cross-compiler do you use? It is not mentioned in the README. I tried to use MXE, with no luck.

daanvV, 2022-07-08:
Thanks for the above info, and FYI I still haven't managed to get this to work. 

@PaulStoffregen I find your suggestion of cross compilation an interesting one. Considering that quite a few windows users are facing issues with this, it would be much appreciated if you could maybe write a short explanation of how to cross-compile for windows? 
Or alternatively maybe put the .exe file that you are able to cross-compile from windows online? (as you [mentioned in this post](https://github.com/PaulStoffregen/teensy_loader_cli/issues/23#issuecomment-215008026))

Many thanks!

ZanzyTHEbar, 2022-07-08:
> Thanks for the above info, and FYI I still haven't managed to get this to work.
> 
> @PaulStoffregen I find your suggestion of cross compilation an interesting one. Considering that quite a few windows users are facing issues with this, it would be much appreciated if you could maybe write a short explanation of how to cross-compile for windows? Or alternatively maybe put the .exe file that you are able to cross-compile from windows online? (as you [mentioned in this post](https://github.com/PaulStoffregen/teensy_loader_cli/issues/23#issuecomment-215008026)) So that other windows users don't have to go through the process of installing Linux to compile something for Windows.
> 
> Many thanks!

If you still are doing this to get remote development working for the tyeensy 4.0 - i got it working. The newest platformio updater have auto-reset and remote development support for the Teensy boards. I didn't need to cross-compile or replace the current teensy-cli. Just start a remote agent on your Host, setup the remote agent to allow a certain email to connect, on your remote dev machine login to Platformio IDE with the email/account associated with the remote agent and simply open the project. In the `Project Tasks` section there will be a `Remote Development` task that allows you to upload to the remote board and see the serial monitor :)

daanvV, 2022-07-08:
@ZanzyTHEbar , thanks for that tip :) My use case is somewhat different however. I want to be able to compile an Arduino file for a teensy board from a python file (hence why I thought using the command line tool would be useful). 
So no need for uploading, remote development etc. Just plain simple compiling an .ino file from the command line. 
If you have any tips then I am all ears! :D