PaulStoffregen/teensy_loader_cli #77: Device is in use by "dummy" driver ??

steeley, 2022-04-26:
just downloaded and compiled the command line loader ok on Mac OSX 11.6.5
Trying to flash a Teensy4.1 - it works in Teensyduino, and I can upload my MicroPython  firmware using the GUI loader on its own.
Command line:
`teensy_loader_cli --mcu=TEENSY41  -v -w TEENSY41-20220117-v1.18.hex `

Produces this error endlessly repeating:
Device is in use by "dummy" driver

Any clues?

thanks

PaulStoffregen, 2022-06-17:
Looks like you've compiled the libusb code which is meant for Linux.

That libusb stuff worked on very old MacOS (like the early days when Apple named versions after big cats) but almost certainly does not work on modern MacOS.

You need to compile with the USB code for use on MacOS.

steeley, 2022-06-17:
Looks like I edited the makefile for macOS but didn't save it.
Works OK now.
thanks.

closed 2022-06-17 by steeley