PaulStoffregen/teensy_loader_cli #81: Cannot flash Teensy40 from Beaglebone Black

LukaBanovic, 2022-09-09:
I get two issues when trying to flash teensy from BBB:
* when cloning this repo and typing in `make` command, I get the `usb.h missing` [issue](https://github.com/PaulStoffregen/teensy_loader_cli/issues/30)
* if I install teensy_loader_cli in any other way, I get [this issue](https://github.com/PaulStoffregen/teensy_loader_cli/issues/72).

PaulStoffregen, 2022-09-09:
You need to install libusb v0.1, as explained in the comments:
https://github.com/PaulStoffregen/teensy_loader_cli/blob/99082869db86f1f5ff7eef0d45262bc7e674f890/teensy_loader_cli.c#L211

```
/****************************************************************/
/*                                                              */
/*             USB Access - libusb (Linux & FreeBSD)            */
/*                                                              */
/*  Uses libusb v0.1. To install:                               */
/*  - [debian, ubuntu, mint] apt install libusb-dev             */
/*  - [redhat, centos]       yum install libusb-devel           */
/*  - [fedora]               dnf install libusb-devel           */
/*  - [arch linux]           pacman -S libusb-compat            */
/*  - [gentoo]               emerge dev-libs/libusb-compat      */
/*                                                              */
/*  - [freebsd]              seems to be preinstalled           */
/****************************************************************/
```

LukaBanovic, 2022-09-09:
I've got `libusb-0.1-4` installed, still throws the same error

LukaBanovic, 2022-09-09:
after a little playing around with libusb versions and libusb-dev, I got rid of the usb.h issue. Thank you.

closed 2022-09-09 by LukaBanovic