PaulStoffregen/teensy_loader_cli #82: teensy_loader_cli.exe for Windows

gonzzzales, 2022-09-12:
Does anyone have a file teensy_loader_cli.exe with support Teensy 4.1?
I cant use Makefile for Windows, compilation goes wrong(((

Evanok, 2022-10-07:
it it working great for me. You just need to install gcc, make and git bash

lambe@XXXXXX MINGW64 ~/work/repository/xxxxxxx/teensy_loader_cli (master)
$ make OS=WINDOWS CC=gcc
gcc -O2 -Wall -s -DUSE_WIN32 -o teensy_loader_cli.exe teensy_loader_cli.c -lhid -lsetupapi -lwinmm

dyson-larry, 2023-02-08:
I would like to revisit this issue, as nobody has provided a satisfactory guide to set up Windows to allow for using the Make feature to create the teensy_loader_cli.exe

I have installed cygwin64 and installed packages for make and gcc but I am still presented with errors relating to missing header files the c file expects to find:

```
$ make
cc -O2 -Wall -s -DUSE_WIN32 -o teensy_loader_cli.exe teensy_loader_cli.c -lhid -lsetupapi -lwinmm
teensy_loader_cli.c:373:10: fatal error: ddk/hidsdi.h: No such file or directory
 #include <ddk/hidsdi.h>
          ^~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:4: teensy_loader_cli.exe] Error 1
```

Having some simple direction/steps to follow to complete the Make build process would be great but I have yet to find any.

@Evanok you seem to have it working great for you, any chance you can elaborate on your setup and what you are using as your make interface?

Evanok, 2023-02-08:
Did you check this: https://github.com/PaulStoffregen/teensy_loader_cli/issues/55

mcsmark, 2023-11-07:
i installed MSYS2 then used ucrt64 to create the exe.

ctadlock, 2024-03-10:
Can get it from PlatformIO.

https://forum.pjrc.com/index.php?threads/teensy_loader_cli-executable-for-windows.63176/page-2#post-340526

bohaoy, 2024-12-27:
> I would like to revisit this issue, as nobody has provided a satisfactory guide to set up Windows to allow for using the Make feature to create the teensy_loader_cli.exe
>
> I have installed cygwin64 [...]

I downloaded the whole development kit from Microsoft and still ran into this issue.

However I was able to fix it by changing `ddk/hidsdi.h` to `hidsdi.h`. Then I was able to compile to create an exe.

For some reason my Makefile didn't work and I had to run the compilation command straight into the terminal.